BondtechAB/Bondtech-Prusa-Firmware #1: [BUG] Random M112 error

rmilyard, 2020-06-06:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Printer type** -  MK3S
**Printer firmware version**-  3.9.0

**MMU Upgrade** - MMU2S
**MMU upgrade firmware version [1.0.6]

**Describe the bug**
During printing in either single of multi color you will get at random times M112 error.  
**To Reproduce**
Whenever try to print
  
**Expected behavior**
Should print without error and stops.
  
**G-code**
  Please attach a G-code. This will make it easier for us to replicate the error.

**Video**
  Please attach a video. It usually helps to solve the problem.
[image]
[image]

ajatoledo, 2020-06-08:
Running into same issue with MK3s that was previously setup as MMU2s running 3.9.0. Running Bondtech with Slice MM-HT in firmware across 4 machines. Only one impacted is the previous MMU2s machine. M112 is seemingly called for no reason, even not printing. Octoprint terminal says firmware error, heartbeat returns 0.

3d-gussner, 2020-06-08:
You should have a search/look/post this at Prusa-Firmware
https://github.com/prusa3d/Prusa-Firmware/issues?q=is%3Aissue+M112

This is not a Bondtech firmware specific issue.

radzimir, 2023-02-15:
I get M112 reproducible during cold start (from 21°C) when extruder reaches 30°C.
After reset and next try extruder goes to 60°C without problem. 
Hardware: Prusa MK2.5S with original Bondtech.
Firmware: Bondtech 3.12.1. 
It does not happen with original firmware from Prusa.

radzimir, 2023-02-19:
Printing ABS, first layer bed 100°C, next layers 110°C. After switching to second layer after a short time M112 - it would be impossible to reach 110 so quick.

radzimir, 2023-02-22:
I discovered that my bed-cable was cold-soldered, it get completely detached after while. 
It is also soldered to mainboard because  the  connector is long melted down anyway.
After resoldering M112 is harder to reproduce, but it still happens. 
MK2.5S uses only 12V for the bed, do it heats up probably slower than 24V printers.

Work around: I take steel plate off and wait until 40°C, then put it on and it goes up to 100 without problems.