BondtechAB/Bondtech-Prusa-Firmware #3: [BUG]<Enter comprehensive title>

ddubyah, 2020-06-23:
Please, before you create a new bug report, please make sure you searched in open and closed issues and couldn't find anything that matches.

**Mk2.5S** 
**FW390-Build3922-Bondtech-MK25S-RAMBo13a-SliceE_HT-Mosquito**

**Describe the bug**
 Nozzle temperature reads 30C too high.
  
**To Reproduce**
Install the Slice Engineering High Temp thermistor and the HT firmware variant. Turn the printer on. The printer shows the bed temperature at 25C (correct ambient temperature) but the nozzle temperature reads 55C, which is 30C too high. 

(I contacted Slice Eng and they advised I measure the resistance at the thermistor. Resistance shows as 550kOhms at an ambient temperature of 25C. This should be correct resistance at that temp. Thermistor seems to be operating correctly which suggests issue with the firmware)
  
**Expected behavior**
Printer should show nozzle temperature at same ambient temperature as the bed.
 

[image]

digitalentropy, 2020-09-05:
Same problem on a MK2.5S. I am disappointed there hasn't been more traction on this over the months that this issue has been known.

SonyaJane, 2020-09-06:
I'm having this issue too. Is there a fixed version of the firmware?

friebel, 2021-12-22:
Slice writes on their [thermistor product page](https://www.sliceengineering.com/products/thermistor-high-temperature?variant=36948577419426):
> The 450°C Thermistor should not be used with printers that use a RAMBo Mini V1.3a or any older version of this board. Common printers with this board include the Prusa MK2.5s, the Prusa MK2.5, the Prusa MK2s, and the Prusa MK2.

I could not find a reason *why* though. Maybe Bondtech just should not offer firmware for the combination MK2*+SliceHT.