BondtechAB/Bondtech-Prusa-Firmware #9: Bondtech LGX with Shortcut Mosquito XYZ Calibration fail

JEBHUB, 2022-11-08:
Hello, I have a Prusa i3 MK3S+.  After installing the new Bondtech extruder and hotend, I found the correct firmware from the Bondtech website and installed that.  After many failed attempts of the XYZ calibration and adjusting the PINDA Probe, I thought I might have the wrong firmware. The problem is after the 4th calibration point it will come back with "XYZ Calibration Failed".  It will go through the 1st, 2nd and 3rd points no problem but after is does the circle thing on the 4th point, I get the XYZ calibration failed.   I've installed 4 different firmwares with the new Bondtech and no luck in getting the XYZ calibration to pass.  The new Bondtech upgrade was for a printer that was fully functioning prior to this install.  I have made sure nothing is in the way of axis travel and double checked to make sure i had the correct firmware, and I believe I have everything in working order.  Can someone please help?

BladerunnerxRC, 2022-11-14:
Has there been a solution for this? I'm planning on ordering the LGX/ShortCut/Mosquito and would like to know it will work.

JEBHUB, 2022-11-14:
> 

I haven't heard anything yet.

JEBHUB, 2022-11-14:
> Has there been a solution for this? I'm planning on ordering the LGX/ShortCut/Mosquito and would like to know it will work.

I haven't heard anything yet.

slayer12bot, 2023-02-05:
I am having the same issue as you. Are you using the shortcut mosquito setup?

BladerunnerxRC, 2023-02-05:
Do a factory reset before installing the latest FW with the Printer on. Push reset, then immediately hold down the selector knob until the factory reset.  This worked for me